DesconBelgrade/Klimerko #5: BME280

kanazir, 2020-01-26:
Please update Aliexpress link to some other item. The listed item is not BME but BMP which can't measure humidity.

epiller, 2020-01-26:
How did you figure? I just checked the link again and it's the BME280. 
If you were judging by the PCB of the sensor itself, it does contain a marking saying "BMP/BME280" which is meant for resellers to tick with a marker but most of them don't do that. All of the BME280s we used were like that.

kanazir, 2020-01-26:
There is Q&A section and the seller says just that (screenshot in attachment). Also, if you browse through customer reviews, you'll find the customers complain they received the wrong one (page 7 of reviews). Maybe I am wrong but the seller failed to send me the sensor in 9 days so maybe he is not so reliable.

I ordered the part from [another seller](https://www.aliexpress.com/item/32654011852.html) and will let you know once the sensor arrives.

[image]