DesconBelgrade/Klimerko #7: Pictures of Klimerko

misavasic, 2020-02-21:
Hi,
Can someone share pictures of assembly process of Klimerko or at least pictures of final product. I have all parts needed except 3D case but I am not so sure how final assembly should look like.
Thanks in advance

epiller, 2020-02-23:
Sorry about that, I thought the instructions in the README were sufficient.  
You'll see the assembled "base" on the photos below, front and back.
I'll get some proper photos and put them in the documentation soon.
[image] [image]

misavasic, 2020-02-24:
Thanks. Instructions are detailed but as someone said "One Look Is Worth A Thousand Words". My goal is to build it without any soldering if possible. When assembled i will post manual and pictures here.

closed 2020-02-24 by misavasic