DesconBelgrade/Klimerko #8: Small README fix

kanazir, 2020-03-15:
Hi,

on [NodeMCU Preparation](https://github.com/DesconBelgrade/Klimerko#nodemcu-preparation) the README says that PMS7003 sensor should be powered from VIN pin of NodeMCU. But, that pin can't be used to power anything on v3 - VUSB should be used instead.

Can you fix it, please?

epiller, 2020-03-18:
Hey, you most likely have the NodeMCU board with CH340 chip (that's the cheaper one 😄). That board is *off* in many ways, such as being incompatible with breadboards, the VIN issue you're having, the inferior serial-to-usb chip and the incompatibility with the 3D printed enclosure provided in this repo. The one linked in this README is the CP2012-based one, which doesn't have any of these issues.

kanazir, 2020-03-18:
Ah... I had one in my drawer for a long time so, I used it for the project, not thinking about its quality :smiley:
Thanks!

closed 2020-03-18 by kanazir