DesconBelgrade/Klimerko #12: Compiling and connecting via Serial Monitor

dcabro, 2020-11-06:
Executable segment sizes:
IROM   : 284884          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 30608   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
DATA   : 1492  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 7948  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 31136 )         - zeroed variables      (global, static) in RAM/HEAP 
Sketch uses 324932 bytes (31%) of program storage space. Maximum is 1044464 bytes.
Global variables use 40576 bytes (49%) of dynamic memory, leaving 41344 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.8
Serial port COM3
Connecting........_____....._____....._____....._____....._____....._____.....____Traceback (most recent call last):
  File "C:\Users\Darko\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4/tools/upload.py", line 65, in <module>
    esptool.main(cmdline)
  File "C:/Users/Darko/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 2890, in main
    esp.connect(args.before)
  File "C:/Users/Darko/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 483, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

_
Also Serial monitor shows garbage although baud rate is properly set:

[image]

dcabro, 2020-11-06:
When pressing reset, Node MCU blinks the blue led once, and serial monitor shows that garbage. All software up to date.

epiller, 2020-11-10:
Are you sure you're using the latest Arduino version, 1.8.13?

dcabro, 2020-11-10:
Yes, 100% sure.
[image]
[image]

epiller, 2020-11-10:
- Do you have any other Serial ports available?
- What does the back of your NodeMCU board say? It should be marked as "Amica"

dcabro, 2020-11-10:
No,only one, COM3
Not sure which is the "back" side, but one side (where the pins are) is white, and has nothing inscribed 
Other side, with wifi module on it, has "HW-306" in the middle. Don't see "Amica" anywhere. Here is the pic. 

Btw, NodeMCU board was ordered months ago - I followed the link to AE from this project on github (btw it's broken right now). Looking at my order details, the item description was "V3 Wireless module NodeMcu 4M bytes Lua WIFI Internet of Things development board based ESP8266 ESP-12E for arduino CP2102"

dcabro, 2020-11-10:
[image]

epiller, 2020-11-10:
That's really weird. The board should look exactly like this: 
[image]
and like this on the bottom side
[image]  

It seems like you've received a counterfeit NodeMCU for some reason.
- Try lowering the upload baud rate (Tools > Upload Speed) to 9600 and then try to upload the firmware again.
- The gibberish you're seeing in the Serial monitor is probably intelligible once you match its baud rate, so try scrolling through different baud rates in Serial monitor and see if the text starts making sense at some point (could be 9600 or 57600) and post it here if you get any results.

dcabro, 2020-11-10:
Yeah, mine is different from that bottom side, but like I said, the exact one I ordered was linked to your page... 

Anyway, 74880 baud seems to be giving something readable, but it does not progress from here:

 ets Jan  8 2013,rst cause:2, boot mode:(7,2)

epiller, 2020-11-10:
Can you try using 9600 baud rate to upload the firmware onto the device?
If that fails, keep the upload speed at 9600 but disconnect the NodeMCU from your computer, then press and hold the FLASH button on the NodeMCU and then (while holding the button), connect it to your computer and only then release the button, and then try uploading again at the same baud rate.

dcabro, 2020-11-10:
57600 is the lowest speed I have in "upload speed"

epiller, 2020-11-10:
Ah, right, sorry about that. Keep it at 115200 and try the second thing from the previous message.

dcabro, 2020-11-10:
Arduino: 1.8.13 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Executable segment sizes:

IROM   : 284884          - code in flash         (default or ICACHE_FLASH_ATTR) 

IRAM   : 30608   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 

DATA   : 1492  )         - initialized variables (global, static) in RAM/HEAP 

RODATA : 7948  ) / 81920 - constants             (global, static) in RAM/HEAP 

BSS    : 31136 )         - zeroed variables      (global, static) in RAM/HEAP 

Sketch uses 324932 bytes (31%) of program storage space. Maximum is 1044464 bytes.

Global variables use 40576 bytes (49%) of dynamic memory, leaving 41344 bytes for local variables. Maximum is 81920 bytes.

esptool.py v2.8

Serial port COM3

Connecting........_____....._____....._____....._____....._____....._____.....____Traceback (most recent call last):

  File "C:\Users\darko\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4/tools/upload.py", line 65, in <module>

    esptool.main(cmdline)

  File "C:/Users/darko/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 2890, in main

    esp.connect(args.before)

  File "C:/Users/darko/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 483, in connect

    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))

esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

_

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

dcabro, 2020-11-10:
And the serial console is showing: 

```
 ets Jan  8 2013,rst cause:2, boot mode:(7,6)

waiting for host
```

dcabro, 2020-12-04:
In most of the cases, it's the STUPID user (that would be me), and not the hardware or the code...

Anyway, I ordered a new node mcu, and when it arrived, before I re-soldered everything, I wanted to make sure that it was okay in the first place. And you know what? PMS7003 RX was soldered do D8 instead of D6. So when I fixed that, everything started working without issues. 

So sorry for all the trouble and thank you!

closed 2020-12-04 by dcabro