DesconBelgrade/Klimerko #14: Can't connect to AllThingsTalk

abukva, 2020-12-26:
Since two days ago my Klimerko can't connect to AllThingsTalk in order to deposit data. I've did reconfiguration of all parameters, tried configuring new Token and even tried creating a new device and configuring with that. I've turned on debug mode in `Klimerko_Firmware/src/AllThingsTalk/AllThingsTalk_WiFi.h` line ` [195]   bool debugVerboseEnabled = false; ` to `true` and I see that Klimerko manages to send Firmware version asset and wifi-signal, after that it gets stuck on Connecting to AllThingsTalk...... line. 

```
WiFi Signal: Excellent
---------------- HTTP Response from AllThingsTalk (Begin) ----------------
HTTP/1.1 200 OK
Content-Type: application/json; charset=utf-8
Server: Mono-HTTPAPI/1.0
Date: Sat, 26 Dec 2020 09:31:21 GMT
Transfer-Encoding: chunked

1a7
{"deviceId":"MYDEVICEID","thing":null,"id":"rMHJDGjI4NGMa1cPAducjlc4","name":"firmware","title":"Firmware Version","is":"sensor","description":null,"createdOn":"2020-12-05T19:53:06.587Z","updatedOn":"2020-12-26T09:31:21.812Z","createdBy":null,"updatedBy":"rMHJDGjI4NGMa1cPAducjlc4","profile":{"type":"string"},"control":{"name":""},"state":{"at":"2020-12-23T14:00:39.45Z","value":"1.3.0"},"style":"undefined"}
0

----------------- HTTP Response from AllThingsTalk (End) -----------------
---------------- HTTP Response from AllThingsTalk (Begin) ----------------
HTTP/1.1 200 OK
Content-Type: application/json; charset=utf-8
Server: Mono-HTTPAPI/1.0
Date: Sat, 26 Dec 2020 09:31:21 GMT
Transfer-Encoding: chunked

1b3
{"deviceId":"MYDEVICEID","thing":null,"id":"9sXVAxIkZRHecBtmDd95IWMD","name":"wifi-signal","title":"WiFi Signal Strength","is":"sensor","description":null,"createdOn":"2020-12-05T19:53:06.586Z","updatedOn":"2020-12-26T09:31:21.931Z","createdBy":null,"updatedBy":"9sXVAxIkZRHecBtmDd95IWMD","profile":{"type":"string"},"control":{"name":""},"state":{"at":"2020-12-24T07:20:43.289Z","value":"Excellent"},"style":"undefined"}
0

----------------- HTTP Response from AllThingsTalk (End) -----------------
HTTP Connection (for Asset creation) to AllThingsTalk Closed
Connecting to AllThingsTalk...............................................................................................................................
```

epiller, 2020-12-28:
- What model NodeMCU are you using? Please attach a photo along with the wiring.
- What board definition do you have installed? Make sure you're running esp8266 core 2.7.4
- Make **sure** you're using the correct AllThingsTalk Credentials. Check if you're copying the Device ID correctly and make sure you're copying the whole Device Token, including "maker:"
- Check if you're running Arduino IDE 1.8.13

abukva, 2020-12-31:
Hi, it turned at the end that AllThingsTalk randomly blocked my IP, so after restarting my router and acquiring a new IP everything works now.

closed 2020-12-31 by abukva