DesconBelgrade/Klimerko #15: Klimerko stops working - dots in serial monitor

dcabro, 2021-01-04:
Not sure why it stops working, but I only see dots in serial monitor. This has happened at least 5-10 times so far. 

Is there a way to debug this? Or do you know what those dots mean?

epiller, 2021-01-04:
Can you double check your AllThingsTalk credentials? Try generating a new Token if you're sure they're correct.

dcabro, 2021-01-04:
Everything starts working without issues after hard reset (power off/on)

dcabro, 2021-01-09:
So it happened again: 
[image]

dcabro, 2021-01-10:
And again.

epiller, 2021-01-10:
What you're seeing is the device trying to connect to something (the blue LED on the NodeMCU should be fading in/out). I'm assuming you opened Serial Monitor _after_ the device stopped responding and if that's the case, make sure to have Serial Monitor open before this issue happens so you can see exactly what it says before it fails.
Before you do this, I'd recommend changing line **113** `device.debugPort(Serial);` to `device.debugPort(Serial, true);` which will enable verbose debug output from the AllThingsTalk SDK.

dcabro, 2021-01-10:
Correct, I opened the Serial Monitor _after_ the device stopped responding. I've changed the line 113 and will leave the serial monitor open. Will let you know when it stops working again. Thanks.

dcabro, 2021-01-11:
`Asset: wifi-signal, Value: Good
AllThingsTalk Connection Dropped! Reason: Network connection was broken
Connecting to WiFi: darknet.....................................................`

So the wifi connection was dropped. But why doesn't it reconnect?

epiller, 2021-01-12:
Are you running the latest ESP8266 core? If you are, I have no idea what could be causing this as I don't have a way to test it because it works on my end (and seems to work on other people's devices as well).

dcabro, 2021-01-12:
I'm running ESP8266 2.7.4. So do you think it's a hardware issue or a software one? I do have a spare NodeMCU that I can switch to if you think it's a hardware issue.

epiller, 2021-01-12:
It could very well be a software issue, but it's really hard to tell what's up. I'd say go for it and change the board, but before you do that, check your WiFi: Try changing the channel, seeing how many devices are connected to the 2.4GHz network (if it's a dual band router), connecting to a different WiFi, etc.

dcabro, 2021-01-12:
I have other wifi clients (like cell phones) and they are all working fine,
or at least they do not get stuck if they are disconnected.

Furthermore, I don't believe that wifi client that never gets disconnected
from the AP/Router even exists. Instead this probably happens often and its
no big deal as long as they reconnect and continue working.

The real question is which part of the code is responsible for the wifi
connection? I'm gonna ping those devs if that's not you.

epiller, 2021-01-19:
Nothing pops in my mind right now except going deep into ESP Arduino Core WiFi classes.

Milutix, 2021-01-24:
Filename: ATT_ESP8266.h
Line: 292

There is a while loop which never ends.
It prints out the '.' character every 10000ms.

I presume you should be trying to reconnect instead?
Line 291 should be placed within this while loop if I understand this correctly.

epiller, 2021-01-25:
> Filename: ATT_ESP8266.h
> Line: 292
> 
> There is a while loop which never ends.
> It prints out the '.' character every 10000ms.
> 
> I presume you should be trying to reconnect instead?
> Line 291 should be placed within this while loop if I understand this correctly.

It does end - once the device reconnects to WiFi but in your case it doesn't.
The sketch you upload to your ESP8266 is not the sole thing that's going to run on it. There's some additional stuff happening in background, one of them being WiFi connection handling (which is the reason why delay(x) for prolonged periods crashes the connection). 
Once WiFi.begin is executed, credentials are passed to the underlying WiFi handling functions and the device is given permission to start connecting to WiFi. If it ever disconnects, it should automatically try to reconnect again, but this is invisible to the user. 
I will look into this in more detail, try to replicate the issue on my Klimerko and figure out a fix.
@dcabro I saw that thread sometime last year but I'll have to look at it again. Thanks!

dcabro, 2021-02-09:
Pulled the device out of use.

closed 2021-02-09 by dcabro

epiller, 2021-12-15:
Hey @dcabro, the new [2.0.0 release](https://github.com/DesconBelgrade/Klimerko/releases/tag/2.0.0) may fix this issue.