DesconBelgrade/Klimerko #18: Using batery

nexbg, 2021-02-04:
Hi everyone, 

Did anyone used autonomous energy source like battery to supply station with power. I do not have electricity source outside, and trying to find way to make it easy. 

BR,
Nemanja

epiller, 2021-02-04:
Hey, due to the linear power regulation circuit on the NodeMCU and the fact that PMS7003 requires 5V to operate (because of the fan), efficient battery powered operation isn't something that could be done well without creating a custom circuit board. What I suggest is that you use a power bank, but don't expect long runtime - I'd guesstimate around 2-3 days.

batica81, 2021-02-04:
Depends on the bank of course. I tried with a small one with two 18650
cells and that lasted about one day.

nexbg, 2021-02-04:
Thanks!
So, power bank with solar panel can be good solution actually. :)

epiller, 2021-02-04:
Thanks for the input @batica81. I would also like to add that the firmware isn't optimized for battery powered operation, in a sense that NodeMCU isn't going into low-power state at all. This could be implemented but would require a whole bunch of changes, a lot of testing and a hardware change (two pins on the NodeMCU would need to be connected to each other in order to enable waking up from deep sleep).

closed 2021-02-07 by epiller