DesconBelgrade/Klimerko #22: System Error: Air Quality Sensor (PMS7003) returned no data on data request this time.

y0d4a, 2021-03-10:
Zdravo,

imam problem sa dobijanjem data sa senzora za vazduh.
Tacnije error koji dobijam je "System Error: Air Quality Sensor (PMS7003) returned no data on data request this time."
Ono sto sam uspeo da vidim, jeste da VIN na nodemcu daje 1.6V, a sam senzor je 5V.
Sta mogu da uradim da resim ovaj prob.?

hvala

milcha, 2021-03-11:
Ja sam imao isti problem i samo sam zicu sa VIN-a zalemio na jedan od 3.3V i sad radi bez problema....

Pozdav

y0d4a, 2021-03-11:
to i meni pade na pamet, al` reko bolje da pitam, i videh da je senzor na 5V, pa reko dal` ce biti dovoljno...
Ok, probacu, hvala!

epiller, 2021-03-11:
Izgleda da ste kupili pogresan NodeMCU (sa CHG340 cipom) jer je bio jeftiniji par centi 😄 
PMS7003 Senzor zahtjeva 5V, tako da ga spojite na "5V" pin na NodeMCU (ako nema tog pina, posaljite sliku ovdje). 
Senzor na 3v3 djeluje kao da radi jer salje podatke, ali zapravo 5V koristi za ventilator u sebi tako da vrijednosti koje daje nisu tacne jer vazduh ne cirkulise.

y0d4a, 2021-03-11:
[image]
[image]

taj je model.. da budem iskren, nisam ni gledao da je tacan moden :/
po slici za ovaj model, izgleda da moze na VV ?
[image]

epiller, 2021-03-11:
Da, moze na VV pin

y0d4a, 2021-03-11:
Air Quality is Very Polluted (Average: Very Polluted)
PM 1:          75 (Average: 75)
PM 2.5:        116 (Average: 116)
PM 10:         124 (Average: 124)
Temperature:   21.0 (Average: 21)
Humidity:      34.8 (Average: 34)
Pressure:      1006 (Average: 1006)

hvala na pomoci!

closed 2021-03-11 by y0d4a