DesconBelgrade/Klimerko #23: Znakovi u wifi sifri?

y0d4a, 2021-03-11:
Zdravo, imam problem da nakacim klimerka na wifi koji sadrzi npr. ! karakter u sifri..
Jel to by design :) od strane esp-a ili?

epiller, 2021-03-11:
Mozes li malo bolje opisati problem? (npr. U kom dijelu procesa se to desava? Da li ne dozvoljava pri unosu ili se samo na spaja?)

y0d4a, 2021-03-11:
Dozvoli sve unose i posle restarta, nikako da se poveze, samo tacke redja...
`Connecting to WiFi: wifi_iot..............................................................................................................................................................`

Na routeru vidim stalno `E0:98:X:X:X:X@wifi_iot: disconnected, received deauth: sending station leaving (3)`

epiller, 2021-03-11:
- Probaj iskljuciti Klimerka iz struje komplet pa opet ukljuciti nakon 1 min.
- Provjeri da li imas najbovije verzije Klimerko Firmware-a, ESP8266 Core-a i Arduina
- Ako uklonis "!" iz sifre na ruteru, da li se onda spaja?

y0d4a, 2021-03-11:
**moja je greska bila**, ceo taj wifi sam onesposobio zbog nekog testa i nisam vratio na stare vrednosti :)
sve ok radi, zatvaram case.

closed 2021-03-11 by y0d4a