DesconBelgrade/Klimerko #28: Podrska za HomeAssistant API

stefancvetkovic, 2021-08-08:
Pozdrav,
Mozda bi mogli da napravimo jednu custom integraciju ili integraciju od strane HomeAssistant zajednice (ukoliko je odobre) da se izlozi API koji bi se integrisao sa HomeAssistant-om.

Pozdrav..

CorvusCoraxHD, 2021-08-08:
Podržavam. I meni to izgleda kao izuzetno korisno.

epiller, 2021-10-08:
+10

pedjas, 2022-10-30:
+1

Nešto sam gledo, moglo bi da se uradi obrnuto: da se u Klimerka učita ASPHome firmver koji bi omogućio lak pristup senzorima i povezivanje na Home Assistant, a ne ;ini se da bi bilo komplikovanoda sam Klimerko sa takvim firmerom i dalje šalje podatke i direktno na Klimerko sajt.

ASPHome ima mogućnsot da nod šalje http post i get upite gde treba, što bi trebalo da je dovoljno.

Još nemam Klimerka pa ne mogu da isprobam.

stefancvetkovic, 2022-11-01:
> +1
> 
> Nešto sam gledao, moglo bi da se uradi obrnuto: da se u Klimerka učita ASPHome firmver koji bi omogućio lak pristup senzorima i povezivanje na Home Assistant, a ne ;ini se da bi bilo komplikovanoda sam Klimerko sa takvim firmerom i dalje šalje podatke i direktno na Klimerko sajt.
> 
> ASPHome ima mogućnost da nod šalje http post i get upite gde treba, što bi trebalo da je dovoljno.
> 
> Još nemam Klimerka pa ne mogu da isprobam.

Mislis na esphome platformu?
https://esphome.io/

pedjas, 2022-11-01:
> Mislis na esphome platformu? https://esphome.io/

Da. 

Nešto sam eksperimentisao danas, bez problema sam namestio da ES8266 modul koji je na EPSHome šalje podatke paralelno i na udaljeni web servis na Internetu (ne Kimerko, nisam gledao kakav je njegov API, nego sam napravio neki improvizovani koji prima HTTP post sa json strukturom u telu).

pedjas, 2022-11-01:
Hmm, nesto ne mogu da nadjem API dokumentaciju za Klimerko sajt.

pedjas, 2022-11-17:
Odustao sam od Klimerka. Vidim da je serer sa podacima zatvorenog tipa, nije moguce preuzimati pdoatke, pa cak ne moze covek ni da podeli link na odredjenu mernu stanicu.

Ne vidim svrhu baktati se sa tim. Napravio sam da se podaci šalju na drugu stranu, gde su lako dostupni. Čak i ne mora da se pravi custom fimver, sve se lako uradi u ESPHome.

stefancvetkovic, 2022-11-17:
> Odustao sam od Klimerka. Vidim da je serer sa podacima zatvorenog tipa, nije moguce preuzimati pdoatke, pa cak ne moze covek ni da podeli link na odredjenu mernu stanicu.
>
> Ne vidim svrhu baktati se [...]

Koju si integraciju izabrao? openWeather ili nesto drugo?

pedjas, 2022-11-17:
Trenutno sam testirao da šaljem na svoju custom lokaciju, za probu, a još nisam odlučio na koje ću open servise da šaljem.

epiller, 2022-11-17:
[Klimerko.org](https://klimerko.org) vuče podatke sa [AllThingsTalk Maker](https://maker.allthingstalk.com) ground-a "[Vazduh gradjanima](https://maker.allthingstalk.com/ground/3s61Mf0HvqudWXXjv2QzUCnq/devices)", tako da je moguće koristiti ovaj [API poziv](https://api.allthingstalk.io/swagger/ui/index#operation/Devices_GetDevice) kako bi se dobili podaci odredjenog uredjaja.  
Što se tiče lokalnog pristupa, moguće je dodati funkcionalnost da Klimerko otvara endpoint na samom uredjaju koji vraća trenutnu vrijednost PM1, PM2.5, PM10, temperaturu, vlaznost, ili pritisak na lokalnoj mreži pa ga onda dodati kao [RESTful Sensor](https://www.home-assistant.io/integrations/sensor.rest/) na HomeAssistant platformi.

pedjas, 2022-11-17:
Ja sam uradio obrnuto. Uređjaj očita senzore i to prosledi u Home Asistant a istovremeno pošalje i http-om na remote lokaciju. 
Nema nekog smisla da šaljem podatke negde na Internet pada ih onda odatle dovlačim nazad.

epiller, 2022-11-17:
Ovo za API sam spomenuo u slučaju da hoćeš da prikažeš podatke sa tudjeg senzora u svojoj HomeAssistant instanci.

pedjas, 2022-11-17:
> Ovo za API sam spomenuo u slučaju da hoćeš da prikažeš podatke sa tudjeg senzora (npr. u blizini) u svojoj HomeAssistant instanci.

A da, to bi bilo dobro. Ima jedan Klimerko u blizini pa ću probati sa njim. Kako da saznam ID senzora? Na Klimerko sajtu tog podatka nema.

Bilo bi dobro i da se napravi integracija za Home Assistant tako da se u HA prosto doda uređaj i u konfiguraciji upiše ID na AllThingsTalk i da se dobiju podaci. Da ne mora ručno da se parsira.

epiller, 2022-11-17:
Probaj ga preko imena naći na AllThingsTalk Maker-u, i tamo onda možeš vidjeti ID.  
Slažem se da bi bilo dobro ali ne znam ko bi to odradio 😕