DesconBelgrade/Klimerko #29: Exception problem

saidkg, 2021-08-14:
Zdravo,

dobijam exception i restart odmah po kačenju na wifi.
Exception (0):
epc1=0x40206b9e epc2=0x00000000 epc3=0x40101a31 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffcb0 end: 3fffffc0 offset: 0190
3ffffe40:  3fff01f4 3fff018c 3fff1304 40206b9e  
3ffffe50:  00000000 003f003f 00000000 00000000  
3ffffe60:  003f003f 00000000 00000000 0036003f  
3ffffe70:  00000000 00000000 002b002f 00000000  
3ffffe80:  00000000 0024002f 00000000 4020b630  
3ffffe90:  3ffe960e 3fff018c 3ffffed0 4020b661  
3ffffea0:  402096d0 3ffffed0 3ffffed0 402032f8  
3ffffeb0:  00000000 0000746e 00000000 40203094  
3ffffec0:  3ffe9909 3fff145c 3fff018c 40203c98  
3ffffed0:  3fff01e8 3fff01dc 3fff01d0 00000002  
3ffffee0:  3ffe9909 3fff018c 3fff11c0 3ffe92cf  
3ffffef0:  3fff11c0 3fff0f8c 3fff018c 40206d00  
3fffff00:  3fff018c 0000676e 4020f644 40206648  
3fffff10:  00000000 0000726f 00000000 00000000  
3fffff20:  0014001f 00000000 00000000 000b000f  
3fffff30:  00000000 3fff018c 3fffff78 4020b421  
3fffff40:  3ffe9909 3fff018c 3fff14a0 4020230c  
3fffff50:  00000040 00000000 feefeffe feefeffe  
3fffff60:  00000000 6c00676e 00000000 00000000  
3fffff70:  fe00726f 00000000 00000000 0010001f  
3fffff80:  00000000 00000000 65726177 00000000  
3fffff90:  feefeffe feefeffe feefeffe 3fff15e4  
3fffffa0:  3fffdad0 00000000 3fff15d0 4020bfe0  
3fffffb0:  feefeffe feefeffe 3ffe86a0 401014dd  
<<<stack<<<

Dekodirano (valjda) u ovo
Exception 0: Illegal instruction
PC: 0x40206b9e
EXCVADDR: 0x00000000

Decoding stack results
0x4020b630: Stream::SendGenericRegular(Print*, int, unsigned int) at C:\Users\saidk\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\StreamSend.cpp line 257
0x4020b661: Stream::SendGenericRegular(Print*, int, unsigned int) at C:\Users\saidk\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\StreamSend.cpp line 267
0x402096d0: SPIClass::setFrequency(unsigned int) at C:\Users\saidk\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\SPI\SPI.cpp line 274
0x402032f8: Device::send(CborPayload&) at C:\Users\saidk\Downloads\Klimerko-master\Klimerko_Firmware\src\AllThingsTalk/ATT_ESP8266.h line 864
0x40203094: ArduinoJson6114_000001::DynamicStringWriter ::write(unsigned char const*, unsigned int) at C:\Users\saidk\Downloads\Klimerko-master\Klimerko_Firmware\src\AllThingsTalk\ArduinoJson\Serialization/DynamicStringWriter.hpp line 40
0x40203c98: Device::debugVerbose (double, char) at C:\Users\saidk\Downloads\Klimerko-master\Klimerko_Firmware\src\AllThingsTalk/ATT_ESP8266.h line 50
0x40206d00: Device::createAssets() at C:\Users\saidk\Downloads\Klimerko-master\Klimerko_Firmware\src\AllThingsTalk/ATT_ESP8266.h line 421
0x4020f644: DhcpServer::end() at C:\Users\saidk\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\LwipDhcpServer.cpp line 1098
0x40206648: Device::mqttCallback(char*, unsigned char*, unsigned int) at C:\Users\saidk\Downloads\Klimerko-master\Klimerko_Firmware\src\AllThingsTalk/ATT_ESP8266.h line 757
0x4020b421: Stream::findUntil(char const*, unsigned int, char const*, unsigned int) at C:\Users\saidk\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\Stream.cpp line 107
0x4020bfe0: operator+(char const*, String const&) at C:\Users\saidk\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/WString.h line 295
0x401014dd: calloc(size_t, size_t) at C:\Users\saidk\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 1181

saidkg, 2021-08-15:
My bad! It appears I used the wrong board definition...
https://github.com/DesconBelgrade/Klimerko/issues/24#issuecomment-845539105

closed 2021-08-15 by saidkg

epiller, 2021-12-15:
Hey @saidkg, just wanted to notify you the issue is resolved with the [newest 2.0.0 release](https://github.com/DesconBelgrade/Klimerko/releases/tag/2.0.0)