DesconBelgrade/Klimerko #51: Documenting MQTT messages

jrocketfingers, 2022-12-13:
Hi folks, I'm running a Klimerko-like setup (same sensors, same NodeMCU, but possibly ESP32 instead), with ESPHome firmware. **If I understand Klimerko's code correctly**, it's just sending MQTT messages to `api.allthingstalk.io`, and klimerko.org is picking up Klimerko devices from there? If this is the case, I could contribute sensor data as well if I knew the messages being sent.

I should be able to deduce MQTT messages from the codebase, but are there any intentions to document this? If not, I'd be glad to document them if I manage to make it work. In that case, could you direct me on how/where would you want it documented?

jrocketfingers, 2022-12-13:
Just read this off a Klimerko unit UART logs, pasting it here for future reference:
```json
{"air-quality":{"value":"Good"},"pm1":{"value":16},"pm2-5":{"value":22},"pm10":{"value":24},"temperature":{"value":19.59},"humidity":{"value":26.51},"pressure":{"value":1003.7},"firmware":{"value":"2.0.0"},"wifi-signal":{"value":"Good"}}
```

epiller, 2022-12-20:
Hey @jrocketfingers, Klimerko follows the [AllThingsTalk MQTT topic/payload convention documented here](https://docs.allthingstalk.com/developers/api/messaging/).

jrocketfingers, 2022-12-21:
Hey @epiller, thanks! I've seen that doc, but it seems to explain the per-asset messages, not the bulk transfer that Klimerko seems to be making. I assume you've actually used [these device states documented on the data formats page](https://docs.allthingstalk.com/developers/data-formats/#device-states)?

epiller, 2022-12-28:
@jrocketfingers Yup! That's exactly what's being used.

dkwireless, 2023-12-18:
Can you please share config yaml so we dont have to do it all ove again? :)

dkwireless, 2023-12-19:
No problem. Thanks. I'm still waiting on parts. Just ordered them. But will write here if I manage to build something.

dkwireless, 2024-03-01:
Sucess!
Thanks @jrocketfingers for pointing me in right direction.

You can find code on https://github.com/dkwireless/ESPHOME-Klimerko