DesconBelgrade/Klimerko #57: Integrating with Home Assistant

DjordjeMandic, 2023-06-10:
Just a suggestion, when its already on the network mdns discovery should be implemented and home asstiant integration made for it.

psyhomb, 2024-03-10:
You can use existing [RESTful](https://www.home-assistant.io/integrations/sensor.rest) integration to add PM sensors in Home Assistant.

Example:

```yaml
- name: Klimerko PM2.5
  unique_id: klimerko_pm2_5
  platform: rest
  resource: https://api.allthingstalk.io/asset/<ASSET_ID>
  headers:
    Authorization: !secret att_maker_api_token
    Content-Type: application/json
  device_class: pm25
  icon: mdi:molecule
  json_attributes_path: "$.state"
  json_attributes:
    - at
  value_template: "{{ value_json.state.value }}"
```

rajkoc, 2024-03-11:
Has it worked for anyone?

psyhomb, 2024-03-11:
It is working for me.  🙂 
[image]

rajkoc, 2024-03-11:
Solved, it works. In the authorization line, add Bearer and then token.

psyhomb, 2024-03-11:
In my case `Bearer` is part of the secret.

`config/secrets.yaml`
```yaml
att_maker_api_token: "Bearer maker:<TOKEN>"
```

psyhomb, 2024-03-13:
I have also created a more comprehensive configuration that will create and collect data from all Klimerko sensors (PM1, PM2.5, PM10, Temperature, Humidity and Atmospheric Pressure) all in one API call.

You only need to replace `<DEVICE_ID` with real value and to update secret per [this](https://github.com/DesconBelgrade/Klimerko/issues/57#issuecomment-1988473141) post. 

```yaml
rest:
  # Klimerko
  - resource: https://api.allthingstalk.io/device/<DEVICE_ID>
    method: GET
    headers:
      Authorization: !secret att_maker_api_token
      Content-Type: application/json
    scan_interval: 60
    sensor:
      # PM1 sensor
      - name: Klimerko PM1
        unique_id: klimerko_pm1
        device_class: pm1
        unit_of_measurement: "µg/m³"
        icon: mdi:molecule
        json_attributes_path: "$.assets[?(@.name=='pm1')].state"
        json_attributes:
          - at
        value_template: |
          {% set sensor = value_json.assets | selectattr("name", "eq", "pm1") | first | default %}
          {{- sensor.state.value }}
      # PM2.5 sensor
      - name: Klimerko PM2.5
        unique_id: klimerko_pm2_5
        device_class: pm25
        unit_of_measurement: "µg/m³"
        icon: mdi:molecule
        json_attributes_path: "$.assets[?(@.name=='pm2-5')].state"
        json_attributes:
          - at
        value_template: |
          {% set sensor = value_json.assets | selectattr("name", "eq", "pm2-5") | first | default %}
          {{- sensor.state.value }}
      # PM10 sensor
      - name: Klimerko PM10
        unique_id: klimerko_pm10
        device_class: pm10
        unit_of_measurement: "µg/m³"
        icon: mdi:molecule
        json_attributes_path: "$.assets[?(@.name=='pm10')].state"
        json_attributes:
          - at
        value_template: |
          {% set sensor = value_json.assets | selectattr("name", "eq", "pm10") | first | default %}
          {{- sensor.state.value }}
      # Temperature sensor
      - name: Klimerko Temperature
        unique_id: klimerko_temperature
        device_class: temperature
        unit_of_measurement: "°C"
        icon: mdi:thermometer
        json_attributes_path: "$.assets[?(@.name=='temperature')].state"
        json_attributes:
          - at
        value_template: |
          {% set sensor = value_json.assets | selectattr("name", "eq", "temperature") | first | default %}
          {{- sensor.state.value }}
      # Humidity sensor
      - name: Klimerko Humidity
        unique_id: klimerko_humidity
        device_class: humidity
        unit_of_measurement: "%"
        icon: mdi:water-percent
        json_attributes_path: "$.assets[?(@.name=='humidity')].state"
        json_attributes:
          - at
        value_template: |
          {% set sensor = value_json.assets | selectattr("name", "eq", "humidity") | first | default %}
          {{- sensor.state.value }}
      # Atmospheric Pressure sensor
      - name: Klimerko Atmospheric Pressure
        unique_id: klimerko_atmospheric_pressure
        device_class: atmospheric_pressure
        unit_of_measurement: "mbar"
        icon: mdi:gauge
        json_attributes_path: "$.assets[?(@.name=='pressure')].state"
        json_attributes:
          - at
        value_template: |
          {% set sensor = value_json.assets | selectattr("name", "eq", "pressure") | first | default %}
          {{- sensor.state.value }}
```

rajkoc, 2024-03-13:
Thank you very much Miloš. Works great!

DjordjeMandic, 2024-03-13:
This works fine when it has internet access. It's still not a solution to the problem. 

IMO Klimerko should have local integration with home assistant, whole point of smart home is that it works without internet access too.

psyhomb, 2024-03-13:
> This works fine when it has internet access. It's still not a solution to the problem.
> 
> IMO Klimerko should have local integration with home assistant, whole point of smart home is that it works without internet access too.

I couldn't agree more, still at this very moment this is all we got.

Also Klimerko firmware is using pub/sub model (MQTT protocol), thus sensor data are pushed to public broker's endpoint, so one way on how you can generate local data is by provisioning a local broker (e.g. [Mosquitto](https://github.com/home-assistant/addons/blob/master/mosquitto/DOCS.md)) and by modifying Klimerko source code and [default broker endpoint](https://github.com/DesconBelgrade/Klimerko/blob/master/Klimerko_Firmware/Klimerko_Firmware.ino#L51) to use local broker address.
I know, this way you are losing public endpoint but this is just another proposal that you can implement in almost no time...

DjordjeMandic, 2024-03-14:
> So isn't the point of sharing with the community?

It is, my intention was never to move it to local only. My intention is to have it do its job like it is now and besides that publish data to HA also.

My idea was to expose sensors to HA directly via HA API or use one more MQTT connection.

If anyone is willing to contribute we can implement the following:
  - MDNS local discovery (useful for HA integration)
  - Local REST API for direct connections (useful for HA integration)
  - One more MQTT connection server where HA is connected which impelments MQTT Discovery to expose sensors
  
Exposed stuff would include:
  - All sensor data which is already published
  - Current WiFi status (more detailed)

psyhomb, 2024-03-14:
One advantage of [this](https://github.com/DesconBelgrade/Klimerko/issues/57#issuecomment-1994309780) approach is that you can basically add any device that is registered and shared on the AllThingsTalk platform to HomeAssistant, not only devices that you own.
You just need to pick a shared <DEVICE_ID> from the Vazduh Gradjanima joined ground and to use ground token instead of device token (personal).
[image]

psyhomb, 2024-03-14:
> It is, my intention was never to move it to local only. My intention is to have it do its job like it is now and besides that publish data to HA also.
>
> My idea was to expose sensors to HA directly [...]

It would be great to have all these features implemented, the only concern I have, not saying it's justifiable, is that I don't know if this hardware (ESP8266 board) have sufficient amount of resources to support all that and how well and reliable it will work.
This is more of a question then concern, because I really don't have any extended knowledge nor experience with this hardware.

DjordjeMandic, 2024-03-15:
> > > So isn't the point of sharing with the community?
> > 
> > It is, my intention was never to move it to local only. My intention is to have it do its job like it is now and besides that publish data to HA also.
> > 
> > My idea was to expose sensors to HA directly via HA API or use one more MQTT connection.
> > 
> > If anyone is willing to contribute we can implement the following:
> > 
> >     * MDNS local discovery (useful for HA integration)
> > 
> >     * Local REST API for direct connections (useful for HA integration)
> > 
> >     * One more MQTT connection server where HA is connected which impelments MQTT Discovery to expose sensors
> > 
> > 
> > Exposed stuff would include:
> > 
> >     * All sensor data which is already published
> > 
> >     * Current WiFi status (more detailed)
> 
> It would be great to have all these features implemented, the only concern I have, not saying it's justifiable, is that I don't know if this hardware (ESP8266 board) has sufficient amount of resources to support all that and how well and reliable it will work.
> This is more of a question then concern, because I really don't have extended knowledge nor experience with this hardware.

RAM is limit here. ArduinoJson will be biggest issue but it all can be managed with proper memory allocation and code layout to prevent fragmentation as much as possible.

Rest can be managed with freertos (should have been from the start imo when there is multiple services running).

psyhomb, 2024-03-15:
@DjordjeMandic thank you for detailed explanation.

psyhomb, 2024-10-06:
If anyone is interested, I’ve also created a [mini-graph-card](https://github.com/kalkih/mini-graph-card) that can be used with Klimerko [entities](https://github.com/DesconBelgrade/Klimerko/issues/57#issuecomment-1994309780).

```yaml
type: vertical-stack
cards:
  - type: custom:mini-graph-card
    name: Outdoor
    entities:
      - entity: sensor.klimerko_temperature
        name: Temperature
        show_state: true
        state_adaptive_color: true
        aggregate_func: max
      - entity: sensor.klimerko_humidity
        name: Humidity
        show_state: true
        state_adaptive_color: true
        show_line: false
        show_points: true
        show_legend: true
        color: "#2AD4FF"
        y_axis: secondary
        aggregate_func: avg
    color_thresholds:
      - value: -40
        color: "#2AD4FF"
      - value: 16
        color: "#FFA500"
      - value: 29
        color: "#E53935"
    line_width: 4
    font_size_header: 13
    font_size: 70
    hours_to_show: 24
    points_per_hour: 3
    hour24: true
    animate: true
    show:
      icon: false
      fill: true
      labels: true
      labels_secondary: true
  - type: custom:mini-graph-card
    name: Outdoor PM10
    entities:
      - entity: sensor.klimerko_pm10
        show_state: true
        state_adaptive_color: true
    font_size_header: 13
    font_size: 70
    aggregate_func: avg
    decimals: 0
    hours_to_show: 24
    points_per_hour: 2
    line_width: 4
    color_thresholds:
      - value: 0
        color: "#17a355"
      - value: 54
        color: "#e9d109"
      - value: 154
        color: "#ff8c00"
      - value: 254
        color: "#ea270d"
      - value: 354
        color: "#7c2c85"
      - value: 424
        color: "#66001f"
    animate: true
    hour24: true
    show:
      graph: bar
      labels: true
      average: true
      extrema: true
      icon: false
  - type: custom:mini-graph-card
    name: Outdoor PM2.5
    entities:
      - entity: sensor.klimerko_pm2_5
        show_state: true
        state_adaptive_color: true
    font_size_header: 13
    font_size: 70
    aggregate_func: avg
    decimals: 0
    hours_to_show: 24
    points_per_hour: 2
    line_width: 4
    color_thresholds:
      - value: 0
        color: "#17a355"
      - value: 12
        color: "#e9d109"
      - value: 35
        color: "#ff8c00"
      - value: 55
        color: "#ea270d"
      - value: 150
        color: "#7c2c85"
      - value: 250
        color: "#66001f"
    animate: true
    hour24: true
    show:
      graph: bar
      labels: true
      average: true
      extrema: true
      icon: false
```

The final result will look something like this:

[image]