DesconBelgrade/Klimerko #63: Air Quality Sensor (PMS7003) i   Temperature/Humidity/Pressure Sensor (BME280)  stalno su off

ray00sd, 2024-06-24:
Pozdrav vec nekoliko dana pokusavam da ozivim klimerka..
dakle..koristim NDU ESP8266, senzore koji su navedeni u prospektu, pratio sam uputstvo sa tri NDU ploce i konstantno dobijam odgovor 
------------------------------DATA------------------------------
[PMS] Air Quality Sensor (PMS7003) returned no data on data request this time.
[PMS] Air Quality Sensor (PMS7003) seems to be offline!
[BME] Temperature/Humidity/Pressure Sensor (BME280) returned no data this time.
[BME] Temperature/Humidity/Pressure Sensor (BME280) seems to be offline!

Sve je spojeno kako treba, BME280 imam sa 6 pinova u upotrebi su 4 prema uputstvu..
Koristim linux OS i dozvolio sam root upis preko usb porta..

Prelemio nekoliko puta konektrore, radio restart, flash...itd..nemam vise ideja i molim za sugestije..

ravenns, 2024-09-10:
Pozdrav,
Konekcija PM senzora na Vin nije dobra ako koristis USB konektor za napajanje uredjaja. Ako je to tvoj slucaj, onda treba koristiti Vu pin na kojem je 5V.

Sto se tice BM280 senzora, ni meni ne radi. Cudno se ponasa. Imam isto onaj sa 6 pinova. Pokusao sam da ga prikacim na sva 3 pina na kojima je 3V ali nece da radi (ne ocitava vrfednost). Plus je na jednom od tih pinova u koliziji sa PM senzorom koji zaglupi ako koristim taj 3V pin (sto nema smisla). Mislim da je moj BM280 neispravan, pa cu probati drugi.

ravenns, 2024-09-10:
Sada sam proverio, ovaj moj BME280 je ustvari BMP280.... Proveri pod mikroskopom, ako imas ili macro na kameri telefona. Ako vidis oznaku da pocinje sa "K" onda imas BMP280...

ravenns, 2024-09-30:
Pozdrav,
I ja se isto patim sa PMS7003 koji nasumicno ocitava. Od 10 puta, jednom ocita, ali vrednosti su nula. Ja koristim ESP8266 sa CH340.

Buljubasa, 2024-12-03:
Moj problem je druge prirode, u senzorima koji se razlikuju od onih pre nekog vremena.
Malo analizirajuci rad video sam da je problem sto se ne dize iz sleep moda, kada se podigne sve radi normalno.
Pokusavajuci svasta  na kraju je upalilo sto sam sledecu liniju:

    pms.wakeUp();

...Ponovio tako da ih ima ukupno pet zaredom.
Nije neko elegantno resenje ali sada radi ocitavanje PMS senzora svaki put i ostavicu tako dok ne dodje neka zvanicna zakrpa.
Razgovarao sam sa autorima, znaju za problem i izbacice novu verziju u skorije vreme.